darwinbeing/HPServerPSUHack #3: PL30

xrayos, 2023-09-16:
do you have a high resolution image of the soldering points for the resistor to increase the voltage? also, which resistor did you use in your final product that you use for coding the bmw. I'm trying to do the same thing. thanks!

darwinbeing, 2023-09-16:
### HERE IS MY PL30 OUTPUT 14.4V FOR BMW CODING SETUP
OUTPUT: 1.27K(TO GROUND)
OVP: 4.7K(TO GROUND)
[image]

xrayos, 2023-09-16:
thank you! I'm going to try and patch firmware for OVP

Which are you using for ground to change the output to 14.4v?

[image]

darwinbeing, 2023-09-16:
Same as the markings in the diagram you provided.

xrayos, 2023-09-16:
Thanks!

what about the 2 potentiometers at the top of the board? can they be use to change the voltage?

darwinbeing, 2023-09-17:
Using a potentiometer would make it more convenient to adjust the output.

xrayos, 2023-09-17:
what about these two by the arrows? Can they be used to adjust the voltage? Thanks!

[image]

darwinbeing, 2023-09-17:
No, they cannot.

closed 2023-09-20 by darwinbeing

cotfas, 2025-02-05:
I am also interested about those two adjustable resistors (potentiometers) at the top.

I have the PL30 and I want to know what are those two pots for?

I think my SMPS is current limited, as I can't draw more than 15amps from it? (I think).