darwinbeing/HPServerPSUHack #7: HSTNS-PD 19

pan-sattan, 2024-03-03:
Hi,

quick question... when I add the resistors as on your pictures... will it work? Or should be modified somehow the software?

Thank you for clarification

darwinbeing, 2024-03-04:
If you set the power supply output voltage to 14.4V, use resistor values as follows: OVP1: approximately 17k, OVP2: 22k, OUTPUT: 22k.
If you set the power supply output voltage to 14V, use resistor values as follows: OVP1: 22k, OVP2: 28k, OUTPUT: 28k.
Attention: Please note that the resistor paralleled with OVP1 is prone to lifting pads during soldering. Therefore, replacing it with a 0603-size resistor would make soldering easier.

pan-sattan, 2024-03-04:
thank you for answer... I need something about 14,8V. On my shop floor is present hot air gun and special equipment... I prepared  OVP1 12K OVP2 22K and Output 22K

darwinbeing, 2024-03-08:
To achieve a 14.8V output with the resistor values
OVP1 ~ 15.5kΩ
OVP2 ~ 19.2kΩ
OUTPUT ~ 19.2kΩ

pan-sattan, 2024-03-10:
> If you set the power supply output voltage to 14.4V, use resistor values as follows: OVP1: approximately 17k, OVP2: 22k, OUTPUT: 22k. If you set the power supply output voltage to 14V, use resistor [...]

I am just preparing components... and thinking.. do you mean to replace whole resistor placed on board for OVP1 17k?? when I replace it, will it be OK? I found only 16,9K 0603 resistor... 

concider that 14,4V will be good enough for me

[image]

darwinbeing, 2024-03-10:
Yes, both options should work. You can either parallel another resistor to achieve a value close to 16.9k, or you can replace it with a 3k 0603 resistor.

[YAGEO Through Hole 16.9K MFR-25FBF52-16K9](https://www.mouser.com/ProductDetail/YAGEO/MFR-25FBF52-16K9?qs=oAGoVhmvjhwdC5bv3mq7Dw%3D%3D)
[YAGEO 0603 3K RT0603DRD073KL](https://www.mouser.com/ProductDetail/YAGEO/RT0603DRD073KL?qs=4CyHz7GXqSgg8wnv2%2FaYDw%3D%3D)

closed 2024-03-13 by darwinbeing