darwinbeing/HPServerPSUHack #9: Vout and OVP for HSTNS-PF43

ogni846, 2024-08-30:
[image]

darwinbeing, 2024-09-03:
PMP9622
350W Constant Voltage - Constant Current (CVCC) Phase-Shifted Full Bridge reference design
https://www.ti.com/lit/df/tidrde5/tidrde5.pdf

ogni846, 2024-09-03:
Now I've put PF43 aside for now, there's not enough material for reworking. I bought PL18 rev15, I'll try it on it3 сент.

https://www.ti.com/lit/df/tidrde5/tidrde5.pdf

darwinbeing, 2024-09-06:
[image]
This can be done using CC-CV control using two feedback loops; one two regulate the maximum output voltage and the other to regulate the maximum output current. The two control loops are OR'd together. The control loop with the lowest error "wins" and controls the output.

IceShiva, 2024-09-07:
Dont forget about the complex poles and zeros in compensation. Of course you can use +1uf cap  integrator in current loop. Psu would be very slow but will works. Second way is use mcu in the loop, extract parameters from analog voltage loop, implement it in digital way and tune it, or open the loop and design new one. Most of server psu's has a current share loop which is not acceptable for charnig batteries and powering leds, because it's implemented as kind of constant resistance, so if impedance of reciver changes current grows up then you need to adjust reference for current share loop. Another limitation is CCM/DCM mode and frequency of operation (Ton/Tof). Good point of start is having oscilloscope, signal generator, dummy load and alot calculations.

darwinbeing, 2024-09-07:
[image]
Thanks for the insights! Regarding hardware modifications, I completely agree that they can introduce instability and unreliability into the power supply. It’s interesting that the HP HSTNS-PL30 doesn’t use the UCC28950 Phase-shifted full-bridge chip and instead implements control in the secondary MCU firmware. Rewriting the secondary firmware could potentially enable constant current mode with minimal hardware changes.

closed 2024-10-09 by darwinbeing