darwinbeing/HPServerPSUHack #11: PL-11 OCP mode problem

NazikSoft, 2024-11-13:
Hi, I have two PL11 (rev 9 and 12). I going to use them as chargers for my battery. I've Increase out voltage up to 14,4v. To brake OVP I used different solution, I've replaced resistor 4,9k (R900 on scheme) to 5,6k. As a result, all works well, but... If I start charging with 14,4v 110a, power supply shutdown, I guess this is some kind of total power protection. I've made your OCP mod, add resistor and have tried to set limit 60A, my BMS show charging current as 60a, but pl11 start buzzing and shutdown in 5-8sec, green led turn off. Looks like power supply is overloaded.

Here my version OVP mode
[image]
[image]

Could you please help me? Does my ovp wrong? Or this ocp is not fit for my situation? Thanks!

darwinbeing, 2024-11-22:
The load is too high, causing the power supply to trigger overcurrent protection. It’s best to use multiple power supplies in parallel to provide a higher output current.

NazikSoft, 2024-11-23:
Can I disable overcurrent protection? Could you tell please, why I have an overcurrent protection when I set output current as 60a? It's only a half of supply power

darwinbeing, 2024-11-23:
You can set the OCP threshold very high, but if the load is too large and the output power exceeds the power supply’s rated capacity, overheating will still cause it to enter protection mode. If your output current is 110A and you set the OCP to 60A, it won’t make the power supply output a constant 60A; it will only trigger OCP earlier. This is not a constant current source and cannot ensure a stable output current. It is recommended to use multiple power supplies in parallel.

NazikSoft, 2024-12-01:
Thank for explanation

closed 2024-12-01 by NazikSoft