darwinbeing/HPServerPSUHack #14: Can i get a lower voltage than 12v?

xiaohu21500, 2024-12-16:
The output voltage depends on Rp, which is less than 220 ohm with a parallel resistance. Can i change R117(220ohm) to a larger value resister to get a lower output voltage?

xiaohu21500, 2024-12-16:
And in issue#1, author mentioned that OVP needs to be modified. is it necessary? can i modify the firmware instead?

darwinbeing, 2024-12-16:
Yes, replacing R117 with a larger resistor will decrease the output voltage. However, if the output voltage becomes too low, such as around 10V, it may trigger UVP (Under Voltage Protection). To address this, you'll also need to adjust the voltage divider resistors for OVP (Over Voltage Protection) or modify the firmware to accommodate the new settings.

xiaohu21500, 2024-12-18:
Thanks, modifying the firmware to close OVP seems to be a more elegant way, however i am not capable to decompile the .hex file. Looking forward to your update！：）
 
One more question,  is there a list of the register address for I2C communication? I believe reading voltage/current can be done with stm32, esp...... which are much cheaper.

IceShiva, 2024-12-18:
> Thanks, modifying the firmware to close OVP seems to be a more elegant way, however i am not capable to decompile the .hex file. Looking forward to your update！：）
>
> One more question, is there a list [...]

Registers needs to be reversed, it's basically uses pmbus with diffrent opcodes. Put your hex file on ida or ghidra and select proper mcu, I did some research how server is interacting with power supply in hp gen8-9 psus by decompiling ilo firmware, psu generally is runtime tuned to specific workload (transcient response) and power profile. Most recent psu uses specialized mcu's for power conversion system and it's fully digital but in most cases reading of firmware is hard or not possible. Hardware in 80+ Titanium psus with bridgeless pfc and resonant halfbridge/fullbridge allows bidirectional current flow, all what need to be done inspect hardware R/L/C of components in power block ( resonant tank if smps is soft switched, transformer ratio and inductance ), mcu board (nowdays these board have few pins (pwm out, sensing and power in)  and write new firmware.  I believe, some peoples in some day would start project's as ev conversion communities began.

xiaohu21500, 2024-12-20:
> Registers needs to be reversed, it's basically uses pmbus with diffrent opcodes. Put your hex file on ida or ghidra and select proper mcu, I did some research how server is interacting with power [...]

OK, it is more complicated than i thought, Your information is important for me to start with. I'd like to share if i make some progress.

closed 2024-12-20 by xiaohu21500