darwinbeing/HPServerPSUHack #16: HSTNS-PL30 external potentiometer

go-0n-git, 2025-02-01:
this is not a issue... i have a question
for  HSTNS-PL30 could you post a detailed picture of where to solder points to add External Voltage 500 ohm potentiometer, and Amp 500 ohm potentiometer

darwinbeing, 2025-02-03:
#6

go-0n-git, 2025-02-04:
i want to add potentiometers to adjust Voltage and current.  could you give detailed pictures location of solder points.

cotfas, 2025-02-04:
I'm also interested on the PL30 to add an adjustable resistor for voltage mod.

From my understanding is to add a resistor (not sure what value), at the pin of the 220 SMD and to GND.

[image]

cotfas, 2025-02-04:
Also, the two potentiometers on the top of the board, I understood it right? They are for adjusting Current ?

But what exactly ?

cotfas, 2025-02-04:
@darwinbeing I have these two videos:

https://www.youtube.com/watch?v=eh7FzcV9E4g&t=817s

https://www.youtube.com/watch?v=CMEkzsVtGsU

Could you tell me what value had you used for adjusting the voltage — on the adjustable resistor pot.

Might it be 10kOhm adjustable resistor? (I'm asking this as I saw from the other video, for the same model, they might used the 10k one)

At the moment, I'm only interested in adjusting the voltage till the OVP kicks in, like in your video, 13.8V should be fine for my amp application.

Thank you!
Vlad

darwinbeing, 2025-02-05:
The trimmer marked with "3" can be rotated counterclockwise to reduce the OCP.

cotfas, 2025-02-06:
[image]

This is what I have.

I did some measurements right now:

The one that is marked with 3 (left side):
—measured both bottom legs of the pot, has = 2,3kOhm
—measured Top + left = 2,3kOhm
—measured Top + Right = 0

The 2nd resistor (marked with 2), from the right:
—measured both bottom legs (the ones from the left side) = 99.3Ohm
—measured left pin and top one(right side) = 0
—measured left pin and right bottom (from the picture, basically it should be the bottom left if resistor would be standup) = 99.3Ohm

@darwinbeing 

1. my assumption is that the OCP is set very low? right?
2. what is the purpose of the right adjustable resistor?
3. how much I should rotate the first resistor (marked with 3) to be in the safe zone? or I also need a dummy load and do tests with it?

darwinbeing, 2025-02-06:
The adjustment range of the left trimmer is limited and safe. The right trimmer is for amplifier input offset voltage compensation—do not adjust it.

cotfas, 2025-02-06:
Thank you for your help!

Also wanted to ask, in your video on YT, for the PL30 version, what adjustment resistor value you were using, for adjusting the output voltage (near the 221 SMD) ?

Is it 10k Ohm adjustable resistor?

cotfas, 2025-02-06:
[image]

https://www.youtube.com/watch?v=eh7FzcV9E4g&t=818s

That blue resistor, what value you were using?

Thank you!

darwinbeing, 2025-02-06:
It's 10k potentiometer.

cotfas, 2025-02-06:
> The adjustment range of the left trimmer is limited and safe. The right trimmer is for amplifier input offset voltage compensation—do not adjust it unless you know what you're doing.

From the schematic that you added, you're talking about the R11 (from the right of the schematic, but from the left of the PCB, marked with 3), that one I can adjust it for OCP? Right?

And R9 I don't touch it.

Correct?

darwinbeing, 2025-02-06:
R11, R12, R13, and R14 make up a voltage divider network. You can adjust them to change the OCP.

cotfas, 2025-02-06:
Perfect! Thank you!

closed 2025-02-06 by darwinbeing