darwinbeing/HPServerPSUHack #17: HSTNS-PL11 with older revision

je1gaa, 2025-02-08:
This is question, not a issue.

I am planning to modify HSTNS-PL11 Rev. 09.
Is it possible to apply FW patch to one with older revision?

darwinbeing, 2025-02-08:
Back it up first. Give it a shot—it should work.

je1gaa, 2025-02-08:
Thanks, I will upload it later.
By the way, is it possible to get 14.8~15.0V from PL11 by FW mod?

darwinbeing, 2025-02-08:
I haven’t really dug into the PL11, so I’m not sure if it’s doable. But the PL30 can definitely do it.

closed 2025-02-09 by darwinbeing

je1gaa, 2025-02-18:
I dumped firmware from my HSTNS-PL11 Rev.9.
As github does not allows .hex file to be uploaded with pull request, upload it with .txt.
[PIC16F883.txt](https://github.com/user-attachments/files/18841622/PIC16F883.txt)

je1gaa, 2025-02-20:
I made pull request to upload this file.