darwinbeing/HPServerPSUHack #20: In-Circuit Serial Programming and Debugging Pinout

fredmeyer, 2025-04-03:
I have a question about the In-Circuit Serial Programming and Debugging pinout.

The pin order specified on the GitHub page does not work correctly on my HSTNS-PD19 REV06F power supply. I am sure that Pin 5 is GND, but in your schematic, Pin 5 appears as +3V VCC.

The following pins are incorrect. Please help me find the correct pin configuration for my power supply. Thank you.

Pinout:
Pin 1 - PGEC
Pin 2 - PGED
Pin 3 - MCLR
Pin 4 - GND
Pin 5 - Vcc (3.3V)

fredmeyer, 2025-04-04:
Pin 5 is GND
Pin 4 is MCLR
Pin 3 is DATA
Pin 2 is CLOCK
Pin 1 is VDD (+5V)

These are the correct pin assignments for the PD19. I’m attaching the original backup of the HEX file.

If you encounter an error during writing, try flashing while the power input is connected.

I’m leaving it here so you can find it later if needed.

[pd19.TXT](https://github.com/user-attachments/files/19605150/pd19.TXT)

darwinbeing, 2025-04-04:
The debug port definition should match the PD30. #2

fredmeyer, 2025-04-07:
Your GitHub page mentions the PL30 pin definitions, but I didn't see anything about PD30. It definitely does not match PL30. The pin definitions for HSTNS-PD19 are as I mentioned above. If you pay attention, in OVP bypass tutorials for the PD19, one leg of the resistors is shown to be soldered to pin number 5. Pin 5 is GND.

fredmeyer, 2025-04-07:
Another thing I’d like to ask is whether the PIC16F877.hex file included for the PD19 in your repo contains the OVP bypass and the modifications to increase the output voltage to 14 volts? Thank you your answer.

ta2vip, 2025-04-24:
My problem is solved. You can close the topic. 

[image]

[image]
[image]

closed 2025-04-24 by darwinbeing