darwinbeing/HPServerPSUHack #21: PL30 Connection issue

BUGBEAR7, 2025-04-04:
Hello 

I am trying to connect it to Pickit programmer but no matter what I do it is unable to connect.
I have put 10K resistor on pin 1 and pin 2 on Pickit side. I have soldered 5 wire directly to the connector , I have checked and re checked the connection is good and there is not short while soldering. 
PL30       PICKIT
Pin1        Pin5
Pin2        Pin4
Pin3        Pin1
Pin4        Pin3
Pin5        Pin2

Tried while PSU is on , tried when it not connected to power
[image]
[image]

[image]
[image]

No matter what I do when I select Midrange/1.8v Min. It always shows "No Device Detected". Please help me what am I doing wrong

darwinbeing, 2025-04-04:
Have you tried MPLAB X IDE v6.05? It might help.

BUGBEAR7, 2025-04-04:
That is bit complicated for me , tried but I guess that is not working either

[image]

Does that 10K resistor have to be connected all the time. Is there anything have to do with putting in debugging mode or something. 
I do not understand what I am doing wrong with PICKIT programmer.
Does mains supply have to connected while doing this ?
Do I have to hold any specific buttons while turning it on?

darwinbeing, 2025-04-04:
You don't need the 10K resistor; you should use MPLAB IPE to update the firmware.

BUGBEAR7, 2025-04-04:
Thanks a lot 
Just did it via MPLAB IPE

[image]

closed 2025-04-04 by BUGBEAR7