darwinbeing/HPServerPSUHack #23: PL30/42  output of 14.1V firmware request

ninja-zx11, 2025-05-03:
So grateful to you for sharing your fantastic work!!! Successfully uploaded your patched firmware to my PL30 and PL42 with PICkit3 and they are outputting 14V.

May I request the patched firmware for PL30/42  that outputs 14.1 Volts.Want to squeeze out little more amps by going down on the volts and I got AGM battery on my BMW E70 and don't want to over charge on super warm days by going over 14.1volts.

Also is it possible to include Over current protection in the firmware so that it shuts down at around 900~950watts output with 110v input to protect itself? I noticed with the patched firmware, voltage starts dropping to 13 volts at around 80~90 amps but power supply didn't shut down to protect itself.

darwinbeing, 2025-05-20:
The Trimmer Potentiometer (left side) marked with "3" can be rotated counterclockwise to reduce the OCP. #16

ninja-zx11, 2025-05-20:
Many Thanks!! I will try that.  Do you think its possible to have a firmware for 14.1 volts. If its too much or not possible, then I will go ahead with the hardware mod.

ninja-zx11, 2025-05-20:
Awesome!! I will try it soon and will let you know.

ninja-zx11, 2025-05-21:
Success!!  Reprogrammed both PL30 and PL42 and both are outputting 14.02V. Tested it with Fluke87V.  I then moved the OCP pot to max counterclockwise and both are shutting around at 74Amps !!. Perfect!!!  Many thanks my friend!!  Please close this issue.

closed 2025-05-21 by darwinbeing