darwinbeing/HPServerPSUHack #25: Firmware for PL41

robots, 2025-09-14:
Hello,

I have dumped PL41 firmware. But i am not 100%sure which constants are OVP, OCP, and setpoints. It also seems that the firmware is duplicated - same firmware twice in memory for some reason. I would appreciate if you could shed some light on the internal workings you have discovered :) 

[hstns-pl41.zip](https://github.com/user-attachments/files/22322869/hstns-pl41.zip)

IceShiva, 2025-09-14:
It's a dual image layout technology when firmware is updated and someting wrong mcu boots from second image from flash layout, if you didn't correct memory regions defaulted by ghidra/ida dissasemblers you cannot get contants at all because memory secions are not mapped. Guys I saw your's yt shorts about achieving cc mode. I stopped testing this because I was lack of hardware needed for solid test. I stopped on creating switched dummy load for testing transcient response meanwhile studying lot of TI/keysight/Christophe Basso literature and almost get in madness dc is easy ac is hell. Yeah some guy modded 6kw (artesyn) cisco psu to cv/cc power supply but he get rid completelly orginal compensator and did completelly new one but damn so slow and I think it's excludes it for everyday use. So keypoint is decipher compensator values whatever is done (digital/analog) way. Regarding to @darwinbeing shorts some liteons has comensator done using noninverting configuration and I'm not sure that your's one has too, but if yes it very vulnerable to "compensator windup" so when you'll get eg 10amps and you'll want to go higher example 20amps compensator have tendence to windup to max value and this is caused by slow response/ bad gains in compensator.

darwinbeing, 2025-09-14:
I’d really appreciate it if you could share a few clear photos of the secondary control board — the one with the MCU and possibly the phase-shift full-bridge controller. That would help a lot with identifying the circuitry. On similar supplies, AN0, AN1, AN2 and AN3 respectively sample the VCOMP voltage, the secondary-side shunt current, the over-voltage protection signal, and the primary-side current.

IceShiva, 2025-09-14:
Dayumn! That explains stability! I'm not sure that I have some pl30 but whatever I'm ordering few one for test! In mail notifications i saw typical c code definitions. Do you rewriting firmware? What do you think about -48vdc power supplies (on output, not input) eg N77-AC-3KW-V01 I got one and it seems to still hybrid solution where 2 UCC2895 backed by atmel mcu (microchip one) but I believe newer one is also dspic powered. I'm currentelly looking for way to get orginal c2000 programmer in reasonable price because many power related devices is running on that platform (obc from ev, blcd drivers) including server power supplies even from other manufacturers. I also found some schematics from older liteon power supplies like pl11 seems to be already in repo I also got PS-2112-1D-LF  1100/1400w version schematics. I'm really sad because I lost somewhere full spreadsheet from liteon about this power supply including dell and liteon notes and fault analysis. There is any way to contribute in this project? Sorry for being chaotic but I have so many questions. Did you try charge lto/li ion batteries? They have much lowest impedance? Regards.